pedvide/ADC #7: Compile error

mauriceAlthoff, 2015-06-09:
I can't get the examples to compile with arduino studio 1.6.1. I get this error messages for the following examples:

analogContinuousDifferentialRead 

analogContinuousDifferentialRead.ino: In function 'void setup()':
analogContinuousDifferentialRead:18: error: 'A13' was not declared in this scope
'A13' was not declared in this scope

analogRead:

Arduino: 1.6.1 (Linux), TD: 1.23, Board: "Teensy LC, Serial, 48 MHz, US English"

/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp: In constructor 'RingBufferDMA::RingBufferDMA(uint8_t, uint8_t)':
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:57:20: error: 'DMA_TCD0_CSR' was not declared in this scope
DMA_TCD_CSR = &DMA_TCD0_CSR + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:59:22: error: 'DMA_TCD0_SADDR' was not declared in this scope
DMA_TCD_SADDR = &DMA_TCD0_SADDR + dma_offset_32*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:60:21: error: 'DMA_TCD0_SOFF' was not declared in this scope
DMA_TCD_SOFF = &DMA_TCD0_SOFF + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:61:22: error: 'DMA_TCD0_SLAST' was not declared in this scope
DMA_TCD_SLAST = &DMA_TCD0_SLAST + dma_offset_32*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:63:22: error: 'DMA_TCD0_DADDR' was not declared in this scope
DMA_TCD_DADDR = &DMA_TCD0_DADDR + dma_offset_32*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:64:21: error: 'DMA_TCD0_DOFF' was not declared in this scope
DMA_TCD_DOFF = &DMA_TCD0_DOFF + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:65:25: error: 'DMA_TCD0_DLASTSGA' was not declared in this scope
DMA_TCD_DLASTSGA = &DMA_TCD0_DLASTSGA + dma_offset_32*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:67:21: error: 'DMA_TCD0_ATTR' was not declared in this scope
DMA_TCD_ATTR = &DMA_TCD0_ATTR + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:68:28: error: 'DMA_TCD0_NBYTES_MLNO' was not declared in this scope
DMA_TCD_NBYTES_MLNO = &DMA_TCD0_NBYTES_MLNO + dma_offset_32*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:70:30: error: 'DMA_TCD0_CITER_ELINKNO' was not declared in this scope
DMA_TCD_CITER_ELINKNO = &DMA_TCD0_CITER_ELINKNO + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:71:30: error: 'DMA_TCD0_BITER_ELINKNO' was not declared in this scope
DMA_TCD_BITER_ELINKNO = &DMA_TCD0_BITER_ELINKNO + dma_offset_16*DMA_channel;
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp: In member function 'void RingBufferDMA::start()':
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:86:5: error: 'DMA_CR' was not declared in this scope
DMA_CR = 0; // normal mode of operation
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:90:40: error: 'DMA_TCD_ATTR_SIZE_16BIT' was not declared in this scope
*DMA_TCD_ATTR = DMA_TCD_ATTR_SSIZE(DMA_TCD_ATTR_SIZE_16BIT) |
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:90:63: error: 'DMA_TCD_ATTR_SSIZE' was not declared in this scope
*DMA_TCD_ATTR = DMA_TCD_ATTR_SSIZE(DMA_TCD_ATTR_SIZE_16BIT) |
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:91:61: error: 'DMA_TCD_ATTR_DSIZE' was not declared in this scope
DMA_TCD_ATTR_DSIZE(DMA_TCD_ATTR_SIZE_16BIT) |
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:92:38: error: 'DMA_TCD_ATTR_DMOD' was not declared in this scope
DMA_TCD_ATTR_DMOD(4); // src and dst data is 16 bit (2 bytes), buffer size 2^^4 bytes = 8 values
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:106:20: error: 'DMA_TCD_CSR_INTMAJOR' was not declared in this scope
*DMA_TCD_CSR = DMA_TCD_CSR_INTMAJOR; // Control and status: interrupt when major counter is complete
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:108:5: error: 'DMA_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // clear all past request
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:108:41: error: 'DMA_CERQ_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // clear all past request
[... 75 lines cut ...]
*DMA_TCD_CSR = DMA_TCD_CSR_INTMAJOR; // Control and status: interrupt when major counter is complete
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:108:5: error: 'DMA_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // clear all past request
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:108:41: error: 'DMA_CERQ_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // clear all past request
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:109:5: error: 'DMA_CINT' was not declared in this scope
DMA_CINT = DMA_channel; // clear interrupts
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:117:5: error: 'DMA_SERQ' was not declared in this scope
DMA_SERQ = DMA_SERQ_SERQ(DMA_channel); // enable DMA request
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:117:41: error: 'DMA_SERQ_SERQ' was not declared in this scope
DMA_SERQ = DMA_SERQ_SERQ(DMA_channel); // enable DMA request
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp: In destructor 'RingBufferDMA::~RingBufferDMA()':
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:127:5: error: 'DMA_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // stop channel 3
^
/arduino-1.6.1/hardware/teensy/avr/libraries/ADC/RingBufferDMA.cpp:127:41: error: 'DMA_CERQ_CERQ' was not declared in this scope
DMA_CERQ = DMA_CERQ_CERQ(DMA_channel); // stop channel 3
^
Error compiling.

pedvide, 2015-06-09:
First, you are using an old version of the library. The latest version updated RingBufferDMA, however it still doesn't work with Teensy LC.
Second, Teensy LC has no pin A13, so you can't use it.

mauriceAlthoff, 2015-06-09:
thanks for the hint

pedvide, 2015-06-09:
No problem, I hope I can make RingBufferDMA work with Teensy LC soon!

closed 2015-06-09 by pedvide