pedvide/ADC #10: Inconsistencies in tables sc1a2channelADC0 and 1 for Teensy 3.1

pk960, 2015-08-10:
I think I've found some inconsistencies in file ADD.cpp (the latest version this time) in the table sc1a2channelADC0 for Teensy 3.0 and 3.1:

>#if defined(ADC_TEENSY_3_0) || defined(ADC_TEENSY_3_1)
>const uint8_t ADC::sc1a2channelADC0[]= { // new version, gives directly the pin number
>    34, 0, 0, 36, 23, 14, 20, 21, 16, 17, 0, 0, 19, 18, // 0-13
>    15, 22, 23, 0, 0, 35, 0, 37, // 14-21
>    39, 40, 0, 0, 38, 41, 42, 43, // VREF_OUT, A14, temp. sensor, bandgap, VREFH, VREFL.
>    0 // 31 means disabled, but just in case
>};

Shouldn't the second 23 be a 0 ?
Shouldn't a 0 be inserted between 41 and 42 to have a table with 32 entries ?

And also in the table sc1a2channelADC1 : 

>const uint8_t ADC::sc1a2channelADC1[]= { // new version, gives directly the pin number
>    36, 0, 0, 34, 28, 26, 29, 30, 16, 17, 0, 0, 0, 0, // 0-13. 5a=26, 5b=27, 4b=28, 4a=31
>    0, 0, 0, 0, 39, 37, 0, 0, // 14-21
>    0, 0, 0, 0, 38, 41, 0, 42, // 22-29. VREF_OUT, A14, temp. sensor, bandgap, VREFH, VREFL.
>    43
>};

Shouldn't a 0 be inserted at the end to have a table with 32 entries ?

pk960, 2015-08-13:
In the last version, there is also a led blinking left uncommented at the beginning of ADC_Module::analogRead.

pedvide, 2015-08-14:
I'll try to have a look at it soon!
Thanks!!