pedvide/ADC #19: Teensy 3.6 wrong ADC1_IRQ

sb1413, 2017-01-03:
The ADC1_IRQ is set to 40 but it should be 73 for Teensy 3.5/3.6.
I fixed it in file "ADC_Module.cpp" by:

ADC_Module::ADC_Module(uint8_t ADC_number, const uint8_t* const a_channel2sc1a, const ADC_NLIST* const a_diff_table) :
        ADC_num(ADC_number)
         , channel2sc1a(a_channel2sc1a)

.......

, PDB0_CHnC1(&PDB0_CH0C1 + ADC_num*0xA)
#if ADC_NUM_ADCS==2
        , IRQ_ADC(IRQ_ADC0 + ADC_num*(IRQ_ADC1 - IRQ_ADC0))				// fix by SB (Teensy 3.6)
#else
	, IRQ_ADC(IRQ_ADC0 + ADC_num*1)
#endif

KurtE, 2017-02-11:
I ran into this issue as well.  The code above did not compile for me.  I changed it to simply be:
```
#if ADC_NUM_ADCS==2
          , IRQ_ADC(ADC_num? IRQ_ADC1 : IRQ_ADC0) // fix by SB (Teensy 3.6)
#else
          , IRQ_ADC(IRQ_ADC0 + ADC_num*1) 
#endif
```

pedvide, 2017-10-08:
Fixed with https://github.com/pedvide/ADC/pull/22

closed 2017-10-08 by pedvide