pedvide/ADC #21: ADC::sc1a2channelADC1 is not correct for T3.5/T3.6

KurtE, 2017-02-11:
It looks like the table was a simple copy of the T3.2 version of the table.  I have not yet also verified if the other table for ADC0 is correct or not, but I believe the correct values for this table is:
```
#elif defined(ADC_TEENSY_3_5) || defined(ADC_TEENSY_3_6)
const uint8_t ADC::sc1a2channelADC1[]= { // new version, gives directly the pin number
    0, 69, 0, 0, 35, 36, 37, 38, 0, 0, 49, 50, 0, 0, // 0-13.
    31, 32, 0, 39, 71, 65, 0, 0, // 14-21
    0, 67, 0, 0, 0, 0, 0, 0, // 22-29.
    0
};
#endif
```

KurtE, 2017-02-12:
Actually the ADC0 is probably not correct either: 
Probably should be:
```#elif defined(ADC_TEENSY_3_5) || defined(ADC_TEENSY_3_6)
const uint8_t ADC::sc1a2channelADC0[]= { // new version, gives directly the pin number
    0, 68, 0, 64, 23, 14, 20, 21, 16, 17, 0, 0, 19, 18, // 0-13
    15, 22, 0, 33, 34, 0, 0, 0, // 14-21
    0, 66, 0, 0, 70, 0, 0, 0, // 22-29
    0 // 31 means disabled, but just in case
};
#endif // defined
```

pedvide, 2017-10-08:
Fixed with https://github.com/pedvide/ADC/pull/22

closed 2017-10-08 by pedvide

rparasram, 2017-11-30:
I validated only 
A3 (0x09 to pin 17), 
A10 (0x00 to pin 64), 
A11 (0x13 to pin 65), 
A12 (0x0e to pin 31)
for ADC1 and updated the array . It now looks like this:

#elif defined(ADC_TEENSY_3_5) || defined(ADC_TEENSY_3_6)
const uint8_t ADC::sc1a2channelADC1[]= { // new version, gives directly the pin number
    64, 69, 0, 0, 35, 36, 37, 38, 0, 17, 49, 50, 0, 0, // 0-13.
    31, 32, 0, 39, 71, 65, 0, 0, // 14-21
    0, 67, 0, 0, 0, 0, 0, 0, // 22-29.
    0
};
#endif