pedvide/ADC #38: Compile error: enum assignment is now an error

dhbaird, 2019-01-24:
Hi! I'm using a newer compiler (https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads/7-2018-q2-update) which now reports this line as an error:

```
In file included from /home/me/arduino/hardware/teensy/avr/libraries/ADC/ADC.h:45:0,
                 from /home/me/arduinotems/src/pickled/pickled.ino:17:
/home/me/arduino/hardware/teensy/avr/libraries/ADC/ADC_Module.h:130:31: error: could not convert 'REF_DEFAULT' from 'ADC_REF_SOURCE' to 'uint8_t {aka unsigned char}'
     REF_3V3 = ADC_REF_SOURCE::REF_DEFAULT, /*!< 3.3 volts */
                               ^~~~~~~~~~~
/home/me/arduino/hardware/teensy/avr/libraries/ADC/ADC_Module.h:131:31: error: could not convert 'REF_ALT' from 'ADC_REF_SOURCE' to 'uint8_t {aka unsigned char}'
     REF_1V2 = ADC_REF_SOURCE::REF_ALT, /*!< 1.2 volts */
                               ^~~~~~~
/home/me/arduino/hardware/teensy/avr/libraries/ADC/ADC_Module.h:132:31: error: could not convert 'REF_DEFAULT' from 'ADC_REF_SOURCE' to 'uint8_t {aka unsigned char}'
     REF_EXT = ADC_REF_SOURCE::REF_DEFAULT, /*!< External VREF */
                               ^~~~~~~~~~~
/home/me/arduino/hardware/teensy/avr/libraries/ADC/ADC_Module.h:133:28: error: could not convert 'REF_NONE' from 'ADC_REF_SOURCE' to 'uint8_t {aka unsigned char}'
     NONE = ADC_REF_SOURCE::REF_NONE // internal, do not use
                            ^~~~~~~~
Error compiling for board Teensy 3.2 / 3.1.
```

Code at:
https://github.com/pedvide/ADC/blob/0f7487dd41faf170e9b3daa711f675d0a3dcb13e/ADC_Module.h#L130

Let me know if I can help further... Thanks!

dhbaird, 2019-01-24:
Submitted #39 which fixes this issue for me.

dhbaird, 2019-02-04:
Hi! Just wondering if @pedvide or someone may be able to take a look at this. Thanks!

dhbaird, 2019-03-11:
Hi @pedvide, any chance you may take a look at this? Thanks!

pedvide, 2019-12-24:
Hey @dhbaird, sorry it took me so long.
I fixed the issue in the teensy4 branch, as I'm adding Teensy 4 functionality. I'll merge it to master in a few days.
Thank for your PR, I'll close it because I've changed the code quite a bit, so it will give me problems when I merge if I accept it now.

closed 2019-12-24 by pedvide