pedvide/ADC #41: Support for Teensy 4

darkstar007, 2019-09-20:
Currently the codebase doesn't support Teensy 4, and just gives the message:

```
 #error "Board not supported!"
  ^
Error compiling for board Teensy 4.0.
```

Thanks,

Matt

KurtE, 2019-12-12:
@darkstar007 and @pedvide ,

Just a quick FYI - We have been seeing lots of people wanting to use ADC.. So a few of us starting hacking up currently a subset of it.

More details up in the thread: https://forum.pjrc.com/threads/58387-Teensy-4-0-which-pins-for-which-ADC

Right now we have it as it's own library ADCL_t4 

Obviously the best thing would be to merge all of this stuff back into this main library, but if you want to take a look at the WIP: https://github.com/KurtE/ADCL_t4

pedvide, 2019-12-12:
Hey KurtE, I'm very sorry I've been away from Teensy and this library for so long. I was meaning to get back to it and add support for Teensy 4.
How about I give you push access to this repo so you can merge whatever changes you want to make it work.
Would you like that? If there are other people that want to collaborate I'm open to add them too.

KurtE, 2019-12-12:
Hi @pedvide, that would probably work, likewise @mjs513 has been doing a lot of the work and testing on the ADCL_t4 code. 

My thoughts are/were - to get a lot of it working in the library I mentioned.  I was also thinking about the approach for merging.  My current thoughts are to: 
Hopefully make the top level objects compatible enough that both T3 and T4 stuff would go into ADC.h/cpp - But that I would keep the T4 stuff separate at the module level.  That is the adc_module.h/cpp would be setup such the .cpp file only outputs code on T3.x/LC, the header file 
also is mainly for the current boards, but at top is a #if for IMXRT... and if so includes a file like: adc_imxrt_module.h and likewise all of the IMXRT module stuff would go into similar name .cpp file that again would only output code on IMXRT... 

i.e. sort of how the wire library is today in Teensyduino....

Thoughts? 

Also I was going to comment in a few other issues, but might as well also mention here.  
Some of my examples are currently different on implementation.   Example I have a DMA test sketch that does DMA on both ADCs,  at first I was trying to use the RingbufferDMA code, but I noticed very little of it actually does anything, and I kept running into issues, like only the first half of buffer worked...

So I updated to use DMA stuff I know that works, I setup a DMA chain of two items each with their own buffer, linked to each other...  And currently have a helper class that is part of the sketch that sets up the DMA and the like....  Was suggested I split this class out, sort of an alternative to your RingBufferDMA code... 

If your interested in seeing that code, it is in the sketch: https://github.com/KurtE/ADCL_t4/blob/master/Examples/T4_ADC_DMA_RMS/T4_ADC_DMA_RMS.ino

pedvide, 2019-12-24:
Hey @KurtE , checkout the "teensy4" branch... :)
It sort of works, but I need to check everything thoroughly.
I had a look at your code to speed up writing mine, so thanks a lot!
My plans are to make sure everything works and post it in the forum.
Then I want to make the library more efficient, especially for the older and smaller boards.

KurtE, 2019-12-24:
Sounds good. 

FYI - I also currently have a version, that I stated testing today as well:
https://github.com/KurtE/ADC/tree/T4

It also has a different DMA buffer class that I have used now on my test programs with the ADCL_t4

I have also been working with @mjs513

Wibbing49, 2019-12-24:
Hola Pedro,

Do You know a possibility to read the internal temperatursensor in a non blocking mode?
I tried with T3.6 but with no success.

Only analogRead() brings a result, but this makes my longest isr 5us longer.

Buen Natale

Lothar
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

KurtE, 2019-12-24:
@Wibbing49 - Does not sound like a Teeny 4 issue.  Note: temperature on T4 I believe does not use ADC system... 

@pedvide  - I am doing some testing of your stuff and looks like you have a lot of it here.  

Note: For DMA stuff, I found the ring buffer DMA stuff did not work for me.  And for the most part there is nothing but init stuff... 

So here is a sample sketch with DMA.  I had this as a class in our other branch... But included it as two files in sketch, that sets up two buffers for each ADC (optionally both), and links the two, and lets you know when each one completes...  The sketch tries to compute an RMS like value for the dma data. 
I tried running it on T4, T3.6, 3.2. Compiles on T3.5, but need to dig one out.  T-LC needs work...

Let me know what you think.
[T4_ADC_DMA_RMS-191224a.zip](https://github.com/pedvide/ADC/files/3998815/T4_ADC_DMA_RMS-191224a.zip)

Also maybe have a discussion, but analog reads at a specific frequency. 

With T3.x you have the PDB stuff.  With the T4, I have sketch(s) that setup to use QTimer to do the similar thing.  For T4.0 (and this next year another T4.x), it is a little more complex as you need to use the Timer,  You use the XBAR system to route the timer interrupt to the ADC_ETC system, you then setup the ADC_ETC with some magic to say hardware trigger and point back some data, and the actual pins channel is plugged into the ADC_ETC structures.... 

I have example of a version that is working with my ADCL_t4 version, which is up on the other thread. 
Will work on how to simplify setup.   Also questions on use only one timer for both ADCs, or optionally use 2nd one....  

But assuming you will want to integrate in, question is would it be yet another set of timing methods, like the PDB ones and/or should there be a more generic set of apis, like:
startReadsAtFrequency() or startTimerReads, or??? some such name, where different boards internally would rout to use PDB or QTimer.

KurtE, 2019-12-24:
@pedvide, @mjs513  - Likewise on functionality, 

There is also some other interesting functionality on the T4.  While some of its functionality is a lot more limited as compared to some of the T3.x boards, It does have some other interesting functionality, where you can chain analog reads, in the hardware.  
Where when the hardware finishes one conversion, it can automatically startup the conversion of another channel...

I know that @mjs513 had a sketch up on the pjrc forum thread with examples of this.

pedvide, 2019-12-24:
@KurtE, @mjs513. I'm almost done with the core ADC T4 functionality. I managed to have one codebase for all boards, with differences between T4 and the others in the smallest number of places.

Indeed, the whole DMA, RingBuffer, etc was really frustrating for me already in the older boards and I never manged to make it work right. I'm happy if we can integrate your code into the library, especially if it can work with all boards. If you want you can be "in charge" of this (I'll create a new issue to discuss DMA stuff).
The same for the PDB, and equivalent for T4, I'll open a new issue and see what we can do.

I'll focus on testing all boards with all settings for the next day or so (happy christmas btw!). Then I want to have a look at code size and number of instructions, especially for LC and T3.0

darkstar007, 2019-12-28:
Hi All,

I finally got time got time to play with @KurtE code (I'm just about to change jobs and country....) and I got it to work fine. The only issue I ran into is that @pedvide  original ADC code just assumed the name of the ADC ISR, while the @KurtE code it has to be set explicitly - I much prefer it to be set explicitly.

Many thanks to both @KurtE and @pedvide.

Matt

pedvide, 2019-12-29:
I have merged all changes needed to support Teensy 4 to the master branch after quite some testing.
I'll close this issue, feel free to open other issues if you have problems.
Thank you all for your help.

closed 2019-12-29 by pedvide