pedvide/ADC #42: DMA functionality

pedvide, 2019-12-24:
DMA can be used to trigger and transfer conversions from/to the ADC.
The current DMA and RingBuffer code does not work very well.
@KurtE has some working code we need to review and incorporate.

The goal is to have support for DMA transfers from/to the ADC for all boards.

pedvide, 2019-12-24:
See also PR #40, it has some code we may want to use.

KurtE, 2019-12-25:
Here in PR #44 is an example DMA buffer class and example app, that appears to be working as a starting point
I would like additional functionality like the ability to decide if I want to use 1 or 2 buffers.  Should it automatically continue or only fill the buffer and then stop until I restart...

KurtE, 2020-01-04:
@pedvide and @mjs513 - I put up some DMA operations up in the fork/branch: 
https://github.com/KurtE/ADC/tree/T4_Timer

This has support for QTimer on T4.  

It also has my Analog Dma Buffer support in it.  With some extensions that allow you to only pass in one buffer.  In that case when you complete the transfer, it will stop.  You can then call a member function to restart the operation again...  

The branch has a few different examples.  I am not sure if this branch has an example yet of DMA only without the timer.  I believe I put one in the other branch I mentioned, that I ran on several different boards.  I should probably update it as well to show one shot support... 

You might want to take a look to see if some of this looks reasonable.  I will probably do a bit more cleanup.