pedvide/ADC #47: enableInterrupts API changed: adc->enableInterrupts(adc0_isr, ADC_0);

KurtE, 2019-12-31:
Many of the examples still have in them calls like:
```
adc->enableInterrupts(adc0_isr, ADC_0);
adc->enableInterrupts(adc1_isr, ADC_1);
```
Which does not work properly as you now have another parameter for Interrupt priority. 
So for example the second call below still sets interrupts on ADC_0 at priority 1.

pedvide, 2020-01-08:
You are right! Thanks for finding this!
I have changed the ADC API (PR #49 ) to delete all methods that require the adc_num. Those methods should be accessed directly with adc->adcX->method(...), instead of adc->method(..., adc_num).

Methods like analogRead can be used with either syntax because if you don´t provide the adc_num they will find the right adc to use anyways.

pedvide, 2020-01-18:
Fixed with PR #49. Thanks!

closed 2020-01-18 by pedvide