pedvide/ADC #51: Recent interface format changes

kdharbert, 2020-03-24:
The version of the library has broken some of my code. Most of the issue has been traced to the interface changes mentioned here:
https://forum.pjrc.com/threads/59874-Teensyduino-1-51-errors-on-compile-1-48-does-not-ADC-class-implicated

However, this doesn't solve all the recent problems. I'm presently seeing issues getting measurements to complete.  Questions:

This the line I'm using to start conversions. I never calls the interrupt and if I manually read the output after conversions should have been complete the result is always zero.
ADC_ERROR =adc->startSynchronizedContinuous(ANA_PIN0,ANA_PIN1);

I tried this line for a single channel, and it did the same. Interrupt is never called.
  //ADC_ERROR =adc->adc0->startContinuous(ANA_PIN0);

I tried this and the output was -70000.
  adc0=adc->adc0->analogRead(ANA_PIN0);
  Serial.println(adc0);

Obviously something else changed, but I can't tell what.
Is this the correct format to implement the enable interrupts command:
adc-> adc0-> enableInterrupts(adc0_isr);

KurtE, 2020-04-06:
@pedvide  - There has been several people who have posted up on PJRC forum about their programs no longer compile.  

I understand you desire to simplify it, but wondering if it would make sense to add back in those methods which took a ADC num as a parameter, and maybe mark them as depreciated? 

I have not played much with GCC to know the best way to mark them.

pedvide, 2020-04-11:
Hey, yeah it makes sense, I should have done this. I think I can add the methods with a attribute: `[[deprecated("Use the adcX-> method instead")]]`

closed 2020-04-11 by pedvide