pedvide/ADC #52: enablePGA doesn't work on second adc

paulhayes, 2020-04-07:
When using continuous differencial reading on adc1 I get the error 
```Error ADC1: Wrong pin```
But only if I try to enable the PGA. If I omit the call to enablePGA it works, and it works if using adc0. Is this expected?

```
  adc->adc1->enablePGA(1);
  adc->adc1->setAveraging(32); // set number of averages
  adc->adc1->setResolution(16); // set bits of resolution
  adc->adc1->setConversionSpeed(ADC_CONVERSION_SPEED::LOW_SPEED); // change the conversion speed
  adc->adc1->setSamplingSpeed(ADC_SAMPLING_SPEED::LOW_SPEED); // change the sampling speed
  
  adc->adc1->enableInterrupts(adc1_isr);
  adc->startContinuousDifferential(A10,A11,ADC_1);
```

pedvide, 2020-04-11:
What board is this?

paulhayes, 2020-04-11:
Sorry this was on a Teensy 3.2

EclipseBoom, 2020-05-22:
Unless I'm reading the RefMan wrong PGA0 only connects to ADC0.  So even though you can get differential inputs for ADC1 on these pins  (9,10) you can't connect the PGA0 to ADC1.

If that is true then the "wrong pin" error is correct.

I think that's right...

paulhayes, 2020-05-22:
ah thanks. I didn't know.

pedvide, 2020-05-27:
Sorry, for the delay, yeah, that's correct. I think the image on the forum post shows this.

closed 2020-05-27 by pedvide