pedvide/ADC #62: T_4.1 DMA example

Defragster, 2020-06-21:
forum post : https://forum.pjrc.com/threads/61451-Teensyduino-1-53-Beta-2?p=243846&viewfull=1#post243846

Also note: TD 1.53 is in Beta - is ADC ready to get an update in TeensyDuino? Short time frame with new IDE 1.8.13 release.

The example : \libraries\ADC\examples\ adc_dma
uses unsupported setcommands:
----
//    adc->setAveraging(8, ADC_1); // set number of averages
//    adc->setResolution(12, ADC_1); // set bits of resolution
    adc->adc1->setAveraging(8);
    adc->adc1->setResolution(12);

Defragster, 2020-06-21:
Updated post - it is working when example is edited without pulled current github - but the example func() usage above has issue as noted?

pedvide, 2020-06-25:
The example is fine in the master branch.
I'm not sure what version of the library that code is from, but now it's fixed as far as I can see.