pedvide/ADC #71: undefined reference to `AnalogBufferDMA::init(ADC*, signed char)' in PlatformIO Release

Ic-ks, 2021-01-09:
The current Teensy_ADC PlatformIO library version (8.0.38) throws an error when compiling the `adc_dma` example:

`undefined reference to AnalogBufferDMA::init(ADC*, signed char) `

 I noticed that this error is already fixed in the master branch (`#ifdef ADC_USE_DMA` must be moved below the includes in `AnalogBufferDMA.h` and `AnalogBufferDMA.cpp`). 
Would it be possible to release a new PlatformIO library with this fix included?

pedvide, 2021-01-12:
Thanks for the bug report, I have published a new version on platformio. Please let me know if it works!

Ic-ks, 2021-01-13:
Nice! Version: 
[image] works like a charm. Thank you very much!

closed 2021-01-15 by pedvide