pedvide/ADC #77: teensy 4 examples not working

jnivard, 2021-05-24:
Hai The examples are not working correctly for teensy 4 
I had to remove //#ifdef ADC_USE_TIMER  in the timer example 
and than it is working ok. Also the other examples have the same problem.
Arduino 1.8.13 Teensyduino, Version 1.53

Mike4U, 2021-07-15:
This problem persists on T4, T4.1 on all ADC DMA examples with new Arduino 1.8.15 Teensyduino, Version 1.54
[https://forum.pjrc.com/threads/64432-Teensy-3-5-ADC_DMA-example-error?p=266054&viewfull=1#post266054](https://forum.pjrc.com/threads/64432-Teensy-3-5-ADC_DMA-example-error?p=266054&viewfull=1#post266054) shows solution except the sketch too needs mod in the same way

pedvide, 2022-11-27:
No, in general it's best to use the latest version of the library, not the one in Teensyduino