pedvide/ADC #78: undefined reference to `AnalogBufferDMA::init(ADC*, signed char)'

jnivard, 2021-05-29:
I try to compile the adc_timer_dma code and recieve the following error. :

C:\Users\Gebruiker\Documents\Arduino\MySketchs\ADC adc_timer_dma_V5/ADCadc_timer_dma_V5.ino:68: undefined reference to `AnalogBufferDMA::init(ADC*, signed char)'
I have tryed all the examples in this forum treed but all have the same result!
https://forum.pjrc.com/threads/62082-Bug-in-Teensyduino-1-53-ADC-examples-not-working?p=280252#post280252

jnivard, 2021-05-29:
solved by replacing 
#include <AnalogBufferDMA.h> by
#include <AnalogBufferDMA.cpp>

It is compiling correctly now

pedvide, 2022-11-27:
Are you using the latest version of this library or the one included with Teensyduino?