pedvide/ADC #85: ADS 7870

TheArtist5152, 2022-09-12:
I am trying to connect this IC in proteus 8 to test theoretically but unable to achieve any output,

Would love if you help me out

TheArtist5152, 2022-09-12:
[image]

It is a A/D converter IC , I am making a tester for this , If it works will test 8051

KurtE, 2022-09-12:
Sorry, I am not @pedvide - but have done some stuff with this library in the past.  Have not done anything for a long time with 
this, so I don't know if he can help with this or not.

But I am also having a hard time understanding how this relates to this library. 
That is this library is specific to provide more advanced support for the ADC sub-systems built into Teensy 3.x and Teensy 4.x microcontrollers. 

And your description shows an external A/D converter chip hooked up to something through SPI?  And you mention proteus 8, which I think is some software package.  So, I don't understand this issue.

If what you are wanting, is something like a Teensy, to read in the Analog information and package the data up some way to send to the PC in some format that the software package understands, or something like that.  I would suggest that you post a message up on an appropriate forum.  For example, the PJRC forum, if you are using a Teensy.   And in the posting try to describe exactly what you are trying to do, and what hardware you are using.

Good luck

pedvide, 2022-11-27:
Sorry, as @KurtE says your question is not related to the Teensy's ADC. Please ask this question in the teensy forum.

closed 2022-11-27 by pedvide