pedvide/ADC #86: Teensy 4.0 Adc timer not accurate

ghostintranslation, 2022-11-12:
When I set a timer on the adc with a frequency of 352800Hz and an interrupt, I rather get an actual frequency of around 336000. So to overcome that I set a higher frequency, but then when I add simple operations in my code like incrementing variables, I see the actual frequency being lower again. 

To measure that I do print an elapsedMicros after a few calls.

To be sure the print is not the culprit I can do the same in an update function of an audio object and here it's always the right timing.

I understand that blocking long execution code might induce such inaccuracy, but I am not doing much here.

The expected behaviour would be that the interrupt is called at the accurate set frequency.

Is it because of the "high" frequency I'm trying to set? What's the limit on Teensy 4.0?

pedvide, 2022-11-27:
If I understand correctly you only see the issue when printing and not when doing something else? printing is a very slow operation, so it makes sense that it blocks the interrupt for too long. Often the best approach is to update a flag in the interrupt function that the main loop will see and do blocking operations there.

ghostintranslation, 2022-11-27:
What I do is actually only incrementing a variable and after 128 increments then I print the elapsed time. So it's printing once every 128 interrupts, and it's also to have better accuracy since I use elapsedMicros and the frequency is too fast to be measured at each interrupt.

If I remember correctly I even tried to print only every 8*128 because at this frequency I should see 2.9ms since the frequency is 8*44100.

And what I measure does not correspond to the expected period.

But if I print at every call of the update function of an audio object then I see the expected period at every print.

I'll test again and share a small reproductible code later this week.

ghostintranslation, 2022-12-18:
I am now unable to reproduce the issue... You can close this thread, I'll come back here if I manage to reproduce.

closed 2023-03-12 by pedvide